pothosware/SoapySDRPlay2 #1: SDRPlay Development Progress

cjcliffe, 2015-09-15:
Starting an issue for current development status and work for shared discussion.

cjcliffe, 2015-09-16:
Getting there.. just need to implement sampling functions and some smart switching for the bandwidth based on rate.

```
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so
cjmacbook:build ccliffe$ SoapySDRUtil --probe="driver=sdrplay"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=sdrplay
mir_sdr version: 1.1
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_2500_Init(2)

mir_sdr_usb_Init()

mir_sdr_usb_USB DLL: Revision 0.1.1

mir_sdr_usb_Init: Timeout expired/failed to establish connection with the device

mir_sdr_2500_Init: mir_sdr_usb_Init() failed
mir_sdr_Init: mir_sdr_2500_Init() Error 1
```

cjcliffe, 2015-09-16:
Segfaults fixed, getting decent --probe results now:

```
cjmacbook:build ccliffe$ SoapySDRUtil --probe="driver=sdrplay"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=sdrplay
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_2500_Init(2)

mir_sdr_usb_Init()

mir_sdr_usb_USB DLL: Revision 0.1.1

mir_sdr_usb_Init: Timeout expired/failed to establish connection with the device

mir_sdr_2500_Init: mir_sdr_usb_Init() failed
mir_sdr_Init: mir_sdr_2500_Init() Error 1

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRPlay
  hardware=SDRPlay
  mir_sdr_version=1.100000

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 0] dB
  Full freq range: [0.1, 2000] MHz
    RF freq range: [0.1, 2000] MHz
  Sample rates: [0.2, 8] MHz
  Filter bandwidths: [0.2, 8] MHz
```

guruofquality, 2015-09-17:
I took a look at the progress. A few comments:

* setupStream() should throw when it doesn't like the configuration. I don't think many of the client codes are checking for NULL as an error.

* getStreamMTU() -- this could return xi_buffer.size() which is the transfer size here it looks like. Implementations may allocate buffers w/ this, so it cant be 0

* FYI, There is SoapySDR::logf(level, format, args...) for for printf style formatting -- if that was something useful that was overlooked.

* readStream() -- it may be easier to deal w/ the remainder like this:
```
//are elements left in the buffer? if not, do a new read
if (this->elemsLeftInBuffer == 0)
{
    //receive into temp buffer
    err = mir_sdr_ReadPacket(&xi[0], &xq[0], &fs, &grc, &rfc, &fsc);
    this->elemsLeftInBuffer = xi_buffer.size();
    this->offsetInfoBuffer = 0;
}

size_t elemsToConvert = std::min(this->elemsLeftInBuffer, numElems);
//conversion loop here, use offsetInfoBuffer for xi_buffer,xq_buffer

//bump variables for next call into readStream
this->elemsLeftInBuffer -= elemsToConvert;
this->offsetInfoBuffer += elemsToConvert;

return elemsToConvert;
```

cjcliffe, 2015-09-18:
Thanks! I'll make the appropriate adjustments, I think it's getting close

cjcliffe, 2015-09-20:
Current progress is that it's functional, still need to do some fiddling with the bandwidth but you can track the progress at https://github.com/cjcliffe/CubicSDR/issues/144 including build instructions for OSX.

closed 2015-10-07 by cjcliffe