pothosware/SoapySDRPlay2 #2: create wiki and update sidebar

guruofquality, 2015-09-23:


Toontje, 2018-11-01:
tmachielsen@Laptop[±|master ?:14 ✗]:~/rx_tools $ SoapySDRUtil --probe="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay
Error probing device: map::at:  key not found

guruofquality, 2018-11-01:
Its a generic std::map error, probably a bug though. Can you get a gdb backtrace?

gdb --args SoapySDRUtil --probe="driver=sdrplay"
in the terminal type run + enter
presumably it will crash
in the terminal type bt + enter

SDRplay, 2018-11-01:
why are you using "driver=sdrplay"?

SoapySDRUtil --probe=sdrplay

or

SoapySDRUtil --find=sdrplay

works fine.

guruofquality, 2018-11-02:
> why are you using "driver=sdrplay"?

The string is parsed as key/value pairs separated by commas and equals, so the driver key unlike the others just tells SoapySDR's factory/discovery to only look at the sdrplay registered functions.

Anyway, find works, but probe instantiates the device and reads a lot of settings and prints out a summary to the console. So something may be untested/uninitialized thats not usually read before its set, or something like that.

A backtrace would probably point to the exact function in question.

SDRplay, 2018-11-02:
I've never used or seen --probe used in any other way than --probe=sdrplay

So why does --probe=sdrplay work? Should SoapySDRUtil not complain that there isn't a key/value pair specified?

guruofquality, 2018-11-02:
> So why does --probe=sdrplay work? Should SoapySDRUtil not complain that there isn't a key/value pair specified?

It just parses it as the key "sdrplay" and value is a "" blank string. There's nothing that cares about key sdrplay, so its just ignored basically. This just means its not filtering out other devices, not really an issue if there is only one device anyhow :-)

Toontje, 2018-11-03:
No gdb here. This is a Mac.

guruofquality, 2018-11-03:
really just looking for a backtrace if possible. so lldb? https://developer.apple.com/library/archive/documentation/IDEs/Conceptual/gdb_to_lldb_transition_guide/document/lldb-command-examples.html

Toontje, 2018-11-04:
```
tmachielsen@Laptop:~ $ lldb -- SoapySDRUtil --probe="driver=sdrplay"
(lldb) target create "SoapySDRUtil"
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/copy.py", line 52, in <module>
    import weakref
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
Current executable set to 'SoapySDRUtil' (x86_64).
(lldb) settings set -- target.run-args  "--probe=driver=sdrplay"
(lldb) bt
error: invalid process
(lldb) run
Process 13027 launched: '/usr/local/bin/SoapySDRUtil' (x86_64)
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay
Error probing device: map::at:  key not found
Process 13027 exited with status = 1 (0x00000001)
(lldb) bt
error: invalid thread
(lldb)
```

Toontje, 2018-11-04:
OK, my problem was this: https://askubuntu.com/questions/981663/python2-7-broken-by-weakref-import-error-please-help

Now:
```
tmachielsen@Laptop[±|master ?:14 ✗]:~/rx_tools $ lldb -- SoapySDRUtil --probe="driver=sdrplay"
(lldb) target create "SoapySDRUtil"
Current executable set to 'SoapySDRUtil' (x86_64).
(lldb) settings set -- target.run-args  "--probe=driver=sdrplay"
(lldb) run
Process 14025 launched: '/usr/local/bin/SoapySDRUtil' (x86_64)
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=1807058E93
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=255

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 48] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 9] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

Process 14025 exited with status = 0 (0x00000000)
```

guruofquality, 2018-11-04:
Cool. However, I'm confused, so probe normally crashes for you like this: `Error probing device: map::at: key not found`, but its fine when run in the debugger?

closed 2019-03-17 by guruofquality