pothosware/SoapySDRPlay2 #8: CubicSDR and SDRPlay - output bandwidth

frspin, 2015-10-16:
I have downloaded CubicSDR, SoapySDR and SoapySDRPlay for testing with my SDRPlay on Ubuntu 14.04.
No problem at all for compile and link and for run it.
But in USB mode and with a 6000 for bandwidth I get audio starting from 800-900 Hz (upper limit is from Fldigi). How can I modify this? A correct bandwidth for PSK is 200-2700 Hz or 200-3000 Hz
See a screenshot of CubicSDR and Fldigi connected by Pulseaudio 
Regards

Franco Spinelli
IW2DHW
[image]

cjcliffe, 2015-10-16:
@frspin does the "Audio Sample Rate" menu help you at all?  In windows I get about ~10+ output sample rates starting at 3khz and ending at 192khz -- I think that might be what you're looking for but I'm not sure what the options will be for your audio device.

You may want to try compiling with JACK audio support (-DUSE_JACK=1 I think) which will let you create a "dummy" device that can be any old arbitrary sample rate you want -- then you can start CubicSDR and Fldigi (if Fldigi supports it) and CubicSDR will lock the audio sample rate to the rate you set.   I don't have the exact instructions handy but could work them up the next time I'm at my linux desktop where I have it working.

frspin, 2015-10-17:
Il 17/10/2015 01:11, Charles J. Cliffe ha scritto:
> @frspin <https://github.com/frspin> does the "Audio Sample Rate" menu
> help you at all? In windows I get about ~10+ output sample rates
> starting at 3khz and ending at 192khz -- I think that might be what
> you're looking for but I'm not sure what the options will be for your
> audio device.
> 

There is a problem with "bandwidth" selection.
Without changing frequency a bandwidth change need to maintain pitch of
selected station.
As you can see in attached screenshot changing from 2800 to 4800 shift
center frequency of received signal from 861 to 777 Hz

In Windows, using SDR-Console, this don't happen. SDR-Console and VC is
using an audio rate of 48000.

Changing audio rate from 48000 to 22050 modify audio frequency. Now with
a bandwidth of 2800 center frequency is at 1030 and increasing bandwidth
to 4800 increase center frequency at 1058.

Using a bandwidth of 2800 Hz there are signals also in lower frequencies
(starting from 200 Hz)

Regards

Franco Spinelli
IW2DHW

cjcliffe, 2016-09-08:
@frspin is this still an issue with the latest CubicSDR? There were several bandwidth and SSB drifting issues resolved leading up to 0.2.0.

frspin, 2016-09-08:
Il 08/09/2016 02:07, Charles J. Cliffe ha scritto:
> @frspin <https://github.com/frspin> is this still an issue with the
> latest CubicSDR? There were several bandwidth and SSB drifting issues
> resolved leading up to 0.2.0.

No. On recent version of CubicSDR I don't see anymore the problem

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2016-09-08:
@frspin thanks for the confirmation; closing this up!

closed 2016-09-08 by cjcliffe