pothosware/SoapySDRPlay2 #9: Update build instructions

dc1rdb, 2015-10-19:
Build instructions in the Wiki should call for a release build
```
cmake ../ -DCMAKE_BUILD_TYPE=Release
```
to improve performance of the module.

guruofquality, 2015-10-19:
I could add the following to the SoapySDRConfig.cmake, and all the modules would get release flags by default. Sounds good? (I did this for PothosConfig, so I'm not sure why it never made it into SoapySDRConfig)

```
########################################################################
# select the release build type by default to get optimization flags
########################################################################
if(NOT CMAKE_BUILD_TYPE)
   set(CMAKE_BUILD_TYPE "Release")
   message(STATUS "Build type not specified: defaulting to release.")
endif(NOT CMAKE_BUILD_TYPE)
set(CMAKE_BUILD_TYPE ${CMAKE_BUILD_TYPE} CACHE STRING "")
```

dc1rdb, 2015-10-19:
Sounds perfect - thanks!

closed 2015-10-19 by dc1rdb