pothosware/SoapySDRPlay2 #14: Slow tuning (device is reinitialized when it shouldn't be)

xmikos, 2016-07-13:
Changing frequency (when doing frequency hopping / sweeping spectrum) is extremely slow. The problem is that device is always reinitialized when changing frequency to a new one if freq. difference is >= rate / 2 (see here: https://github.com/pothosware/SoapySDRPlay/blob/master/Streaming.cpp#L416).

But this condition is wrong, according to PDF at http://www.sdrplay.com/docs/Mirics_SDR_API_Specification.pdf, device needs to be reinitialized only when a frequency band edge is crossed (see frequency allocation table at the end of the PDF). There is no need for `freqDiff < rate / 2.0` condition.

cjcliffe, 2016-07-13:
@xmikos aye, I think this was mostly in response to the earlier SDRPlay libraries where it was a bit less stable and this was a solution -- I'll investigate patching this to only occur at the band edges as suggested -- thanks!

xmikos, 2016-07-13:
I also think current implementation will fail if you change frequency to less than rate / 2 (so reinitialization will not be triggered), but still cross the frequency band edge. So it is not only enhancement, but also bug. But I am not 100% sure, couldn't test it right now...

cjcliffe, 2016-09-08:
@gedger 's fix takes care of this nicely; thanks!

closed 2016-09-08 by cjcliffe