pothosware/SoapySDRPlay2 #22: Use with CubicSDR problems

gedger, 2016-12-29:
looking forward to get this updated version running but it seems that it will only build against the as yet unreleased v0.6 SoapySDR as it uses the new KwargsToString API which was added after the 0.5.4 release. CubicSDR 0.2.0 is also built against SoapySDR 0.5.2 so I presume the only way of getting CubicSDR to use this is to build it from source?

Thanks

Richard

cjcliffe, 2016-12-29:
@gedger yup -- at the moment I'm building CubicSDR against v0.6; and will release some bookmarking test builds with updated soapy mods for OSX, Windows and Linux shortly.

closed 2017-01-19 by cjcliffe

guruofquality, 2017-01-19:
It should be ok now to compile against releases of soapysdr: https://github.com/pothosware/SoapySDRPlay/commit/d7eb5d7b0e462160b4fb2977da3f52d5c228dfaa