pothosware/SoapySDRPlay2 #23: New version error

frspin, 2016-12-29:
With new version of SoapySDRPlay and with SoapySDR from GIT my SDRPlay is not found

This is the output of SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.0-g5e823fc1
API Version: v0.6.0
ABI Version: v0.6-dev
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.6-dev
Module found: /usr/local/lib/SoapySDR/modules0.6-dev/librtlsdrSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6-dev/libsdrPlaySupport.so
Loading modules... done
Available factories...null, rtlsdr, sdrplay, 

And this is the output of SoapySDRUtil --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=2500 rev=0206 serno=B0002P0001 bus=002 port=001 devAvail=0
No devices found!

How can I get it to work?

Regards

Franco Spinelli

gedger, 2016-12-30:
What OS are you running under, I presume some flavour of Linux?

The devAvail=0 looks wrong, it should say devAvail=1

Did the driver install from sdrplay.com go smoothly? Check the driver message log (dmesg) to see that the driver has been installed OK and loading. I think there should also be a udev rules file. My pi3 has

cat /etc/udev/rules.d/66-mirics.rules

#USB devices
SUBSYSTEM=="usb" ATTR{idVendor}=="1df7" ATTR{idProduct}=="2500" MODE:="0666"
SUBSYSTEM=="usb" ATTR{idVendor}=="1df7" ATTR{idProduct}=="3010" MODE:="0666"

Richard

frspin, 2016-12-30:
Il 30/12/2016 16:13, gedger ha scritto:
> What OS are you running under, I presume some flavour of Linux?

Ubuntu 16.04/64 bit up to date

> 
> The devAvail=0 looks wrong, it should say devAvail=1
> 
> Did the driver install from sdrplay.com go smoothly? Check the driver
> message log (dmesg) to see that the driver has been installed OK and
> loading. I think there should also be a udev rules file. My pi3 has

Latest API is installed and is working. All was working, with some
errors as in my issue, with previous version.

And all is working with gr-osmosdr and also with Luaradio

So, I suppose, no problem with API and no problem with UDEV rule:

SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1df7",ATTRS{idProduct}=="2500",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1df7",ATTRS{idProduct}=="3010",MODE:="0666"

Regards

Franco Spinelli
IW2DHW

frspin, 2016-12-31:
I have done some test and solved the problem.
As first test I have modified Registration.cpp forcing rspDevs[0].devAvail to "1". I have only 1 RSP1 and this worked.
After this SDRUtils show my device ad "available" and CubicSDR is working

So I have search for a kernel module using my RSP1 and found that removing msi2500 will do the job. Now SoapySRTUtil show correctly:

`
mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=2500 rev=0206 serno=B0002P0001 bus=002 port=001 devAvail=1
`

After this test the problem was in kernel module but there is no "usb error" as in old post on SDRPlay forum.
And this kernel module don't interferes in correct RSP1 usage.
My device is working with gr-osmosdr and Luaradio (both don't use mir_sdr_getDevices() call) and was working with old SoapySDRPlay

Regards

Franco Spinelli
IW2DHW

SDRplay, 2017-11-19:
Closed due to inactivity to avoid confusion.

closed 2017-11-19 by SDRplay