pothosware/SoapySDRPlay2 #27: Add RSP2 support for Antenna A,B and HiZ

starvald, 2017-04-10:
Really missed not being able to select the ports in gqrx using soapy until i saw a comment on facebook with a fork adding these features.

https://github.com/fcrohas/SoapySDRPlay

can these features be added to the main repo?

Ronkpa, 2017-04-28:
I'm getting lost somewhere. I was able to CMake this new SoapySDRPlay-master and it generated sdrPlaySupport.dll in its build directory. How do I get POTHOSware to load this for my SDRsource so I can use the HiZ antenna port? Thanks.

vsonnier, 2017-08-19:
@Ronkpa The dll module file must be copied in the module directory of the PothosSDR installation,
which should be `c:/Program Files/PothosSDR/lib/SoapySDR/modules0.6` for a recent PothosSDR version.
By now the PothosSDR Windows [Installer](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D) contains the same Mirics API version v2.10 as the most recent SDRPlay provides. 
Be sure to have the [API/HARDWARE DRIVER – VERSION 2.10](http://www.sdrplay.com/downloads/) installed first.
 
I'm closing this issue because thanks to @SDRplay and my own recent contribution, the Antenna support  for RSP2 works as is should be.

closed 2017-08-19 by vsonnier

Ronkpa, 2017-08-29:
Vincent,

Thank you. But I think I messed up somewhere. I had Pothos build 2017.03.12
working with RSP2 Hi-Z antenna with a cobbled-up DLL that I modified and
built somehow, I don't remember. Yesterday I installed the new version of
Pothos build 2017.08.13 and overwrote my old DLL. Now I don't know where to
find a DLL that works for SDRPlay RSP2 Hi-Z antenna, and especially I don't
know which parameters I have to enter in Pothos. I think I cobbled up the
SoapySDRPlay DLL source code before (last April) to make it work with the
Hi-Z antenna. I remember it had something to do with AmPort Select but I
don't remember the details. Anyway, can you tell me where to get the
SoapySDRPlay DLL and what parameters to enter into the Pothos to make my
RSP2 work with the Hi-Z antenna. Thanks.

Really I just want to *use* Pothos as a tool to get my RSP2 working at low
frequency. I don't really want to work on source code and CMake and all
that stuff right now. Is that possible, or do I have to get into the code
to make this work? Please let me know. In the meantime I'll be hacking away
at it. I'm getting old and my memory is not what it used to be.

Thanks again for your help.
Ron Krahe

vsonnier, 2017-08-30:
@Ronkpa Hello ! Either you wait for the next PothosSDR distrib release, or you can try my molac version comiled with Visual Studio 2017 and put it in the PothosSDR modules dir:
[sdrPlaySupport_master_win64_VS2017.zip](https://github.com/pothosware/SoapySDRPlay/files/1262437/sdrPlaySupport_master_win64_VS2017.zip)