pothosware/SoapySDRPlay2 #28: SoapySDRPlay CMake error on Ubuntu 16.04 LTS

garytagg, 2017-05-29:
Hello,
I am getting this CMake error when trying to build SoapySDRPlay

gary@gary-MSI-Wind-Ubuntu:~/SoapySDRPlay/build$ cmake ..
CMake Error at CMakeLists.txt:7 (find_package):
  Could not find a package configuration file provided by "SoapySDR"
  (requested version 0.4.0) with any of the following names:

    SoapySDRConfig.cmake
    soapysdr-config.cmake

  Add the installation prefix of "SoapySDR" to CMAKE_PREFIX_PATH or set
  "SoapySDR_DIR" to a directory containing one of the above files.  If
  "SoapySDR" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/gary/SoapySDRPlay/build/CMakeFiles/CMakeOutput.log".
===================
I believe I have followed the instructions listed in the SoapySDR Wiki to the letter.

The directory listing of SoapySDRPlay is:
gary@gary-MSI-Wind-Ubuntu:~/SoapySDRPlay$ ls
build          CMakeLists.txt        LICENSE.txt  Registation.cpp  SoapySDRPlay.hpp
Changelog.txt  FindLibSDRplay.cmake  README.md    Settings.cpp     Streaming.cpp

Any advice on how to proceed?

Thank you,
Gary

guruofquality, 2017-05-30:
Primarily, that error means that the SoapySDRPlay cmake files cant find the cmake development files installed by SoapySDR. Can you share the output of make install for SoapySDR? Usually it will install files to: `/usr/local/share/cmake/SoapySDR/` by default

guruofquality, 2017-05-30:
@garytagg This might be related to this issue: https://github.com/pothosware/pothos/issues/110 So my other suggestion is to move git cloned directories into a subdirectory (not home). It happens to confuse the build in this case. Sorry about that.

garytagg, 2017-05-30:
Thank you for helping. I created a directory /home/gary/SDR and moved SoapySDRPlay under that.  I deleted the existing build directory but cmake still gives the same error, so I don't think this is related to #110.  There is no cmake directory in /usr/local/share.  I downloaded the prebuilt SoapySDR packages as described in this URL.  https://github.com/pothosware/pothos/wiki/Ubuntu.  The only packages not downloaded were the SDR Hardware packages as my hardware is SDRPlay. Do I need to uninstall the prebuilt packages and build SoapySDR manually?

guruofquality, 2017-05-30:
>  I downloaded the prebuilt SoapySDR packages as described in this URL. https://github.com/pothosware/pothos/wiki/Ubuntu

In that case you just need the development files as well:
`sudo apt-get install libsoapysdr-dev`

garytagg, 2017-05-30:
issue solved - thank you

closed 2017-05-30 by garytagg

EEEEEEEEEEEEEEEEEEEIEEEEEEEEEEEEEEEEEE, 2020-03-31:
來吃酒

technicalfiddles, 2020-10-17:
The suggestion above to install the development files solved this same problem for me today. I was following the build instructions here: https://github.com/pothosware/SoapyRTLSDR/wiki
Installing on an RPI4 w Raspbian.