pothosware/SoapySDRPlay2 #29: External dependencies not found Windows Visual Studio 2015

michaelrempel, 2017-08-03:
I am a programmer, but the vagueness of C++ and cmake is not something I deal in often.

I had a warning about exceptions not being handled which I resolved with CMAKE_CXX_FLAGS.

The real problem is 75 unresolved external references.

LNK2001	unresolved external symbol "public: virtual bool __thiscall SoapySDR::Device::getFullDuplex(int,unsigned int)const " (?getFullDuplex@Device@SoapySDR@@UBE_NHI@Z)	sdrPlaySupport	D:\SoapySDR\build\Settings.obj	1	

I presume that either I am missing some files or the cmake isnt treating the hpp file properly. devices.cpp is in the root folder under SoapySDRPlay.hpp but never gets copied or delt with in build as far as I can see.
Any help would be appreciated. Thank you.

guruofquality, 2017-08-03:
Did you also build SoapySDR prior to this? It looks like build just failed to link SaopySDR.lib which is why you see all of the linker errors. But I wouldnt expect that since the SoapySDRPlay cmake shouldnt configure properly when it cant find the development files. Can you share the arguments passed to cmake, or maybe a screenshot of cmake gui, and also its output when configuring? I regularly build [SoapySDRPlay](https://github.com/pothosware/PothosSDR/blob/master/BuildSoapySDR.cmake#L212) and I never had to change any of the flags for vc14

SDRplay, 2017-11-19:
Not specific to SoapySDRPlay - closed to keep open issues relevant.

closed 2017-11-19 by SDRplay