pothosware/SoapySDRPlay2 #36: RSP1A bias-t not working

unixpunk, 2017-11-19:
Hi, new to this type of SDR and to SoapySDR as well, so not sure where this issue would be or if its maybe even a hw issue, but the bias-t option as seen through CubicSDR using SoapySDRPlay (all just built from source today on Ubuntu 14.04) doesn't seem to work.  I'm also not sure if the RF Notch option works, not sure best way to confirm.  Other than this, seems to work great in Cubic.  (I'm new to that as well, came from SDR#, gqrx using rtl-sdr and hackrf.)

Thanks!

######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device 

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=255 1712xxxxxx
  mir_sdr_api_version=2.110000
  mir_sdr_hw_version=2

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * ExtRef Enable - External Reference Control
       [key=extref_ctrl, default=true, type=bool]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: Antenna A, Antenna B, Hi-Z
  Corrections: DC removal
  Full gain range: [0, 47] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 8] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

SDRplay, 2017-11-19:
That's the wrong branch by the looks of it. Currently, the right branch to use is: Rebased_SDRPlay_pull_request_34 whilst it's in testing. Hopefully it will push to master very soon. I've just tested the RSP1A Bias-T with the latest CubicSDR and this branch of SoapySDRPlay and it's working fine.

SDRplay, 2017-11-19:
The test branch has now been merged into the master - so just clone the master again and rebuild it and you should be good.

closed 2017-11-19 by SDRplay

unixpunk, 2017-11-19:
That did it, thanks!