pothosware/SoapySDRPlay2 #38: Error Installing in Raspberry Pi 3

lu9cbl, 2017-12-11:
When i'm installing follow the installation process in the wiki, and execute the "make" command, this error appears. any recomendation?

pi@raspberrypi:~/SoapySDRPlay/build $ make
Scanning dependencies of target sdrPlaySupport
[ 25%] Building CXX object CMakeFiles/sdrPlaySupport.dir/Registation.cpp.o
[ 50%] Building CXX object CMakeFiles/sdrPlaySupport.dir/Settings.cpp.o
[ 75%] Building CXX object CMakeFiles/sdrPlaySupport.dir/Streaming.cpp.o
[100%] Linking CXX shared module libsdrPlaySupport.so
/usr/local/lib/libmirsdrapi-rsp.so: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
CMakeFiles/sdrPlaySupport.dir/build.make:148: recipe for target 'libsdrPlaySupport.so' failed
make[2]: *** [libsdrPlaySupport.so] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/sdrPlaySupport.dir/all' failed
make[1]: *** [CMakeFiles/sdrPlaySupport.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Thanks a lot

lu9cbl, 2017-12-11:
I'vd clone this "git clone https://github.com/pothosware/SoapySDRPlay.git"

SDRplay, 2017-12-11:
Understood, but what API did you download from the sdrplay.com website? The API is not part of SoapySDRPlay - it's the application layer that sits between the hardware and SoapySDRPlay

lu9cbl, 2017-12-11:
I'm using the arm64 version. Sorry for misinterprate the question.

SDRplay, 2017-12-11:
Then you need the Raspberry Pi API from our website (https://www.sdrplay.com/downloads)

lu9cbl, 2017-12-13:
Perfect!!! thanks a lot. Later i´m will write a step by step guide to share with others, and put the link here

closed 2017-12-13 by lu9cbl