pothosware/SoapySDRPlay2 #40: RSP1 weird noise

idl0r, 2018-03-08:
Hey,

my RSP1 works pretty well on Windows but using Soapy/SoapySDRPlay on Linux seems to add some weird noise as well as some weird peaks shown in GQRX as well as CubicSDR, but not on Windows.

Any idea what could cause those?

In the GQRX screenshot you'll better see those ..., like resets? Theres also some noise that's not seen in the waterfall, some crackling. 

The frequency range / bandwidth on the GQRX is also very small compared to the one on HDSDR on Windows. On CubicSDR it looks bigger (see the small hill between 86.2 - ~86.5 on the GQRX one). So I can't see anything before or after that area while I do see way more on Windows. With seeing I mean being able to see peaks when someone is actually sending. This one might be a GQRX issue though.

SDRplay, 2018-03-08:
Without knowing how the RSP is configured it's difficult to be certain, but the GQRX one looks like you've got the wrong IFBW (IF bandwidth) - with too narrow IFBW you'll be attenuating all of the signals outside of it.

On the CubicSDR one, when you get those horizontal lines does it correlate to high CPU load? What processor are you using and what is each core load like?

Without seeing more details, it's difficult to comment any further.

Also when using SoapySDR, make sure debug is disabled (default) and what branch of SoapySDRPlay are you using? What library is the GQRX build using to communicate with the RSP?

idl0r, 2018-03-08:
>Without knowing how the RSP is configured it's difficult to be certain, but the GQRX one looks like you've got the wrong IFBW (IF bandwidth) - with too narrow IFBW you'll be attenuating all of the signals outside of it.

CubicSDR:
Offset 0
Sample Rate 2.048 MHz
RF Gain Select 1
IF Mode Zero-IF
IQ Correction True
AGC Setpoint -30

GQRX:
Input Rate 2000000
Decimation none
Sample Rate 2.000 Msps
Bandwidth 0 (default)
LNB LO 0 MHz
Hardware AGC
AGC Slow

>On the CubicSDR one, when you get those horizontal lines does it correlate to high CPU load? What processor are you using and what is each core load like?
CPU is more or less idle - i7-4790K. I've quite a lot running right now but it's still a total load of 0.4 - 0.5

>Also when using SoapySDR, make sure debug is disabled (default) and what branch of SoapySDRPlay are you using? What library is the GQRX build using to communicate with the RSP?
Latest comit

SoapySDRPlay latest master
GQRX 2.10, it's using SoapySDR as well

With debug being disabled you mean a build-time option or runtime option? If it's disabled by default then it should be that default. I haven't enabled it at least.

SDRplay, 2018-03-08:
2.0 MHz sample rate should select 1.536MHz IFBW and that looks correct on the CubicSDR screenshot - strange that in GQRX it says Bandwidth 0 - that can't be right. Seems like extra config is required there.

So when you see a horizontal line, you are monitoring all 8 cpu loads and none peak high?

Out of interest what version of the API are you using?

idl0r, 2018-03-08:
1.536MHz IFBW looks indeed better. Thanks!

>So when you see a horizontal line, you are monitoring all 8 cpu loads and none peak high?
Yup. htop says max. ~10% each right now

>Out of interest what version of the API are you using?
sdrplay-2.11.1

idl0r, 2018-03-08:
Interesting, when my other programs do some CPU it indeed affects the waterfall quite a lot when one core is around 100% even though there may be 7 more (3 without HT) being more or less idle.
Here's a short recording while CPU being idle. https://drive.google.com/open?id=1nfOchkISXUMD0xV5exufkwhyhLfEqr3B

The test with one core at 100% does make quite a lot noise in the waterfall.

idl0r, 2018-03-09:
On Windows with CubicSDR it looks somewhat more stable but the noise is there as well. I also noticed that when I open a new browser tab or switch those tabs or do some actions it's making noise audio wise. Squelch is actually set high enough and the level doesn't rise in the moment when I can hear it anyway. So squelch level + noise level look ok but I still can hear it in that moment when I do stuff in the browser or other things. With HDSDR it doesn't seem to happen. I'll try gqrx again tomorrow or so to see whether it's Soapy related or just CubicSDR in that case.

SDRplay, 2018-03-09:
I looked at the video - I'm not sure what I should be looking at. The "bouncing" of the spectrum is the AGC. If you don't want the AGC on, you can disable it. Unless I missed what you mean?

closed 2018-06-24 by SDRplay