pothosware/SoapySDRPlay2 #42: API v2.13 : Broken Gain Change callback returns out-of-bounds values. (Win64)

vsonnier, 2018-06-22:
Hello @SDRplay ! 
I tried the API v2.13 with your latest compatibility change with CubicSDR on Win64, but it appears 
the Gain Change Callback (§3.56 page 65) passed to `mir_sdr_ErrT mir_sdr_StreamInit` is broken concerning at least the `gRdB` value.

This value is used in the `getGain` call by CubicSDR to refresh the gain sliders values. The observable behaviour is that the gain slider goes back to 0, and stays here no matter we apply another setting.

Turns out the callback value `gRdB` retreived with `SoapySDRPlay::gr_callback(unsigned int gRdB, unsigned int lnaGRdB)`  may suddenly have completely out-of bounds values (INT_MIN for instance) and no further gain application can restore a normal behaviour.

The callback exposes wrong values either with AGC on or OFF. Suddenly, values will get off-the-charts and never come back again.

As a side note, I noticed inconsistencies in the callback description:
§3.4 Callback function prototypes page 11 describes:
typedef void (*mir_sdr_GainChangeCallback_t)(__unsigned int gRidx__, unsigned int gRdB,  
                                             unsigned int lnaGRdB, void *cbContext); 

while
 
§3.56 Gain Change Callback p65 describes:
   typedef void (*mir_sdr_GainChangeCallback_t)(unsigned int gRdB, unsigned int lnaGRdB, 
                                             void *cbContext);  

consistent with the API signature. 
But it also mentions §2.12:
`2.12 Gain Message ID Enumerated Type 
Used to identify messages passed back in the gRdB field of the gain
typedef enum 
{ 
  mir_sdr_GAIN_MESSAGE_START_ID  = 0x80000000, 
  mir_sdr_ADC_OVERLOAD_DETECTED  = mir_sdr_GAIN_MESSAGE_START_ID
  mir_sdr_ADC_OVERLOAD_CORRECTED = mir_sdr_GAIN_MESSAGE_START_ID
} mir_sdr_GainMessageIdT; 
 `

which may relate to this unknown parameter __gRidx__.

SDRplay, 2018-06-22:
Yep that's the ADC overload detection. I'll add the code to deal with that over the weekend. I'll check the doc as well. Thanks.

SDRplay, 2018-06-24:
Gain callback updated to support ADC overload detection. I've also updated the device detection to include the RSPduo. I've shown how the gain value (as apposed to gain reduction) is obtained in the gain callback. It could be used if required.

closed 2018-06-24 by SDRplay