pothosware/SoapySDRPlay2 #43: SoapySDRUtil doesn't detect SDRPlay RSP2

rich0, 2018-08-09:
I've followed most of the troubleshooting steps I've found online and chatted a bit on freenode but haven't been able to figure out why my RSP2 isn't being enumerated.  I'm using:
SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.1-unknown
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr
Search path: /usr/lib64/SoapySDR/modules0.6
Search path: /usr/local/lib64/SoapySDR/modules0.6
Module found: /usr/lib64/SoapySDR/modules0.6/libsdrPlaySupport.so
Loading modules... done
Available factories...null, sdrplay,

SoapySDRUtil --probe
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=null
  hardware=null

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 0 Rx, 0 Tx
  Timestamps: NO

SoapySDRUtil --check=sdrplay
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Loading modules... done
Checking driver 'sdrplay'... PRESENT

Linux ham1 4.17.14 #1 SMP Thu Aug 9 09:45:34 EDT 2018 x86_64 AMD Ryzen 3 2200G with Radeon Vega Graphics AuthenticAMD GNU/Linux

I'm happy to attach kernel configs or whatever you're interested in (strace/etc), or try patches/etc.  This is on Gentoo, but my kernel is custom, and I packaged soapysdrplay and the sdrplay libraries myself (soapysdr is the repository version).  If I get it working I'll likely add the packages to the main repo.  

I have enabled just about all the USB-related options in the kernel except gadget-related options.  Also, "lsmod | grep msi" returns nothing.  The usb device does show up in /dev/bus/usb with 0666 permissions so the udev rule seems to be working fine.

Is there anything else you can think of that I can do to troubleshoot, or can you think of any kernel-related config settings that might be relevant?

unixpunk, 2018-08-09:
https://github.com/pothosware/SoapySDRPlay - Requires the API from SDRPlay, you have that installed already?  "Get SDR Play driver binaries 'API/HW driver v2.x' (not v3.x) from - http://sdrplay.com/downloads"
HTH
PS - best place for questions like this would be https://groups.google.com/d/forum/pothos-users

rich0, 2018-08-09:
Yes, those drivers are installed.  It wouldn't even build without them.  Happy to post in the group - I was told to log an issue in #pothos on freenode...

guruofquality, 2018-08-09:
@rich0 I was under the impression that SoapySDRUtil --probe was printing the output for the sdr play, its basically just calling into the null device and printing diddly. Ex, if you SoapySDRUtil --probe="driver=sdrplay", I think you will get a error message. That doesnt help much, but its consistently not working, rather than ambiguously instantiating/not enumerating like I thought. So something must be up with the binaries.

rich0, 2018-08-09:
Here is the output of --probe with the driver.  I don't think it is actually enumerating the device, but just displaying the capabilities of the driver itself?

SoapySDRUtil --probe="driver=sdrplay"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=sdrplay

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRPlay
  hardware=SDRPlay
  mir_sdr_version=2.130000

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * Low IF (when available) - Use low IF when available: 0.5MHz SR with 200 and 300khz BW, 1MHz SR with 600kHz BW, 2048kHz SR with 1536kHz BW
       [key=use_low_if, default=true, type=bool]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=2048]
  Antennas: RX
  Corrections: DC removal, DC offset
  Full gain range: [0, 204] dB
    TUNER gain range: [0, 102] dB
    LNAT gain range: [0, 102] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
  Sample rates: 0.222222, 0.333333, 0.428571, 0.5, 0.571429, ..., 8, 9, 10, 11, 12 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

guruofquality, 2018-08-09:
Oh there you go, nice output. @SDRplay what would cause the device instantiation to work, but enumeration shows no devices?

SDRplay, 2018-08-09:
Can I see the output of lsusb - i.e. what does the system think is connected? The command should be SoapySDRUtil --probe=sdrplay or SoapySDRUtil --find=sdrplay

rich0, 2018-08-09:
lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 1df7:3010
Bus 003 Device 002: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 046d:c05a Logitech, Inc. M90/M100 Optical Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

SoapySDRUtil --find=sdrplay
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

No devices found!

SoapySDRUtil --probe=sdrplay
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device sdrplay

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=null
  hardware=null

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 0 Rx, 0 Tx
  Timestamps: NO

SDRplay, 2018-08-09:
Something doesn't seem right there - I would... "sudo service udev restart" and then disconnect and reconnect the RSP2 - then try again please.

rich0, 2018-08-09:
I ran (as root) systemctl restart systemd-udevd   

There was no change in behavior for --find=sdrplay or --probe=sdrplay

SDRplay, 2018-08-09:
and you physically disconnected and reconnected the RSP?

rich0, 2018-08-09:
Yes - other than the output in dmesg nothing changed:

[ 4748.741722] usb 3-3: USB disconnect, device number 4
[ 4751.173600] usb 3-3: new high-speed USB device number 5 using xhci_hcd
[ 4751.300581] usb 3-3: New USB device found, idVendor=1df7, idProduct=3010, bcdDevice= 2.00
[ 4751.300587] usb 3-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0

SDRplay, 2018-08-09:
Actually, one second - that's version 0.6 - that's pretty old - where did you get that from? 0.6 probably only works with 2.11 of the API - have you tried building from the latest published source?

rich0, 2018-08-09:
I'm using the Gentoo package for soapysdr-0.6.1.  It should be using the tagged tarball from github for that version.  Do you want me to build from master?

SDRplay, 2018-08-09:
I'm pretty sure 0.6 will only work with 2.11 API - 2.13 API will require the latest version of SoapySDRPlay master

closed 2018-11-02 by SDRplay