pothosware/SoapySDRPlay2 #46: issues with multiple consumers

fmoessbauer, 2019-01-22:
If one consumer uses a Soapy SDRPlay source, a concurrent call to `SoapySDRUtil --find` returns `No devices found!`.
When using the SoapyRemote server, the device is detected correctly and works with the first client. However when the second client attaches (through the server), both clients break with the following message:
```
[ERROR] SoapyRemote::find(tcp://192.168.2.24:55132) -- transact FAIL: SoapyRPCUnpacker::recv(header) FAIL:
```

guruofquality, 2019-01-22:
Theres just no implementation for multiple consumers in SDRPlay or SoapyRemote. Most of these follow a threading model of one thread/caller per stream. I think its something people would like in the latter FWIW.

starvald, 2019-01-22:
That wording threw me too a while back.

guruofquality, 2019-01-22:
Maybe I misread from the title, each rx stream really needs to have its own consumer. But you should be able to have multiple clients connected to the same device, potentially all doing control stuff or reading, writing different streams. Its not clear why it crashed, and it should probably be something fixed. I can suggest running the server in gdb and sharing the backtrace.

fmoessbauer, 2019-01-23:
> Try this, run the server, and in two different terminals run python -c "import SoapySDR; d = SoapySDR.Device('driver=remote'); import time; time.sleep(10)". They should both be able to connect.

This did work correctly. But what does not work is the following:

1. Start SoapyRemote
2. Connect any Client to the Server (e.g. rtl_433)
3. Call `SoapySDRUtil --find="driver=remote"`

The server then crashes on the `SoapySDRUtil` call with the following backtrace:

```
Thread 14 "SoapySDRServer" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffec818700 (LWP 12849)]
0x00007fffed26bed8 in ?? () from /usr/local/lib/libmirsdrapi-rsp.so.2.13
(gdb) bt
#0  0x00007fffed26bed8 in ?? () from /usr/local/lib/libmirsdrapi-rsp.so.2.13
#1  0x00007fffed2667f3 in ?? () from /usr/local/lib/libmirsdrapi-rsp.so.2.13
#2  0x00007fffed267463 in libusb_handle_events_timeout_completed () from /usr/local/lib/libmirsdrapi-rsp.so.2.13
#3  0x00007fffed25e7ac in eventThread () from /usr/local/lib/libmirsdrapi-rsp.so.2.13
#4  0x00007ffff6b2f6db in start_thread (arg=0x7fffec818700) at pthread_create.c:463
#5  0x00007ffff685888f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

```
SoapySDRUtil --find="driver=remote"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] avahi_service_browser_new() failed: Bad state
[ERROR] SoapyRemote::find(tcp://192.168.2.24:55132) -- transact FAIL: SoapyRPCUnpacker::recv() TIMEOUT
No devices found!
```

Unfortunately the symbols for libmirsdrapi are missing. I tried compiling SoapySDR[Play] and SoapyRemote in debug mode, but I guess the libmirsdrapi is from another source?

guruofquality, 2019-01-23:
Here is a simple command line test case with RTLSDR that works FWIW, it should crash for SDRPlay presumably. The server and client can probably all be the same host:

* On the server: `SoapySDRServer --bind`
* On client1 (rate tester runs the stream in a loop): `SoapySDRUtil --args="driver=remote,remote:driver=rtlsdr,remote:prot=tcp" --rate=1e6 --direction=RX`
* On client2 (find the device while rate testing, no crash): `SoapySDRUtil --find="driver=remote,remote:driver=rtlsdr"`

@SDRplay It looks like some code in findSDRPlay() may not be safe to call when the stream is active and it can cause unexpected crashes. I dont know if its purely an issue of thread safety, or something inside of the mirics driver that isnt checking if usb descriptors are already open, etc. My best guess anyway.

SDRplay, 2019-01-23:
I'm not sure what use case you are trying to do, but you can only have one instance of the API in a given user space, so if you want to access 2 RSPs on the same machine, you will need 2 instances of the API and therefore 2 instances of the server application.

guruofquality, 2019-01-23:
There is only once instance, the instance in the server. But that being said, @fmoessbauer can you try the same tests with and without the args `remote,remote:driver=sdrplay` being passed to the find. It might be the case that if this is the same local machine, the find call talks to the server which is fine, but also directly to the soapysdrplay, which is actually the one that crashes.

> I'm not sure what use case you are trying to do, but you can only have one instance of the API in a given user space, so if you want to access 2 RSPs on the same machine, you will need 2 instances of the API and therefore 2 instances of the server application.

Yea, I remember there was that API limitation with no way to specify the handle and the global variables. I don't think thats the case here. Its one device, but the call to `findSDRPlay()`while sdr play is streaming is making the server crash. I dont know if the call is from another process, or just from another thread in the same server.

So whatever that mechanism is that causes the crash, ideally `findSDRPlay()` would just not return any results rather than segfault things. But that all I can discern from this little back and fourth, and I can make a few guesses and suggest some things to test. Thats about it for my usefulness :-P

SDRplay, 2019-01-23:
The very first sentence in this thread is correct and is what should happen. If a RSP is in use, then a call to SoapySDRUtil --find should return no devices found. A device is only "found" if it a) exists and b) is not in use.

Why there is a segmentation fault I have no idea, but if the intention is to access the same RSP twice or to access 2 RSPs with only one instance of the API (2.13) then this isn't going to work.

fmoessbauer, 2019-01-24:
@guruofquality In short: The server segfaults in both cases. I also tested the scenario where the server and each client runs on a distinct physical machine. In this case it crashes (same stacktrace) too.

```
SoapySDRServer --bind
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.1-g2a7363e4
Server UUID: 8a1cca61-bbc8-157c-8567-6f0ca8c01702
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server...
Connecting to DNS-SD daemon...
[INFO] Avahi version:  (null)
[INFO] Avahi hostname: (null)
[INFO] Avahi domain:   (null)
[INFO] Avahi FQDN:     (null)
[ERROR] avahi_entry_group_new() failed
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:192.168.2.23]:42960)
SoapyServerListener::accept([::ffff:192.168.2.23]:42962)
SoapyServerListener::accept([::ffff:192.168.2.23]:42964)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.2.23]:42966)
SoapyServerListener::accept([::ffff:192.168.2.23]:42972)
SoapyServerListener::accept([::ffff:192.168.2.23]:42974)
Segmentation fault (core dumped)
```

Client 1 (same machine)
```
SoapySDRUtil --args="driver=remote,remote:driver=sdrplay,remote:prot=tcp" --rate=1e6 --direction=RX
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] avahi_service_browser_new() failed: Bad state
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CS16, scaleFactor=32767, mtu=4096, window=44040192)
[INFO] Using format CS16.
[INFO] Server side stream bound to [::ffff:192.168.2.23]:42555
[WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 1024 KiB
[INFO] Configured sender endpoint: dgram=4048 bytes, 1006 elements @ 4 bytes, window=1024 KiB
[INFO] Configured receiver endpoint: dgram=4048 bytes, 1006 elements @ 4 bytes, window=43008 KiB
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 1 Msps
Starting stream loop, press Ctrl+C to exit...
[WARNING] Set thread priority 0.5 failed: Operation not permitted
0.974303 Msps   3.89721 MBps
\[ERROR] SoapyLogAcceptor::handlerLoop(tcp://192.168.2.23:55132) FAIL: SoapyRPCUnpacker::recv(header) FAIL: 
[ERROR] StreamEndpoint::acquireRecv(), FAILED recv() [104: Connection reset by peer]
Unexpected stream error STREAM_ERROR
Error in rate test: SoapyRPCUnpacker::recv(header) FAIL:
[ERROR] ~SoapyRemoteDevice() FAIL: SoapyRPCPacker::send() FAIL: send() [32: Broken pipe]
[ERROR] SoapyLogAcceptor::connect(tcp://192.168.2.23:55132) FAIL: connect(tcp://192.168.2.23:55132) [111: Connection refused]
[ERROR] SoapyLogAcceptor::shutdown(tcp://192.168.2.23:55132) FAIL: SoapyRPCPacker::send() FAIL: send() [32: Broken pipe]
```

Client two (same machine):
```
SoapySDRUtil --find="driver=remote,remote:driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] avahi_service_browser_new() failed: Bad state
[ERROR] SoapyLogAcceptor::handlerLoop(tcp://192.168.2.23:55132) FAIL: SoapyRPCUnpacker::recv(header) FAIL: 
[ERROR] SoapyLogAcceptor::connect(tcp://192.168.2.23:55132) FAIL: connect(tcp://192.168.2.23:55132) [111: Connection refused]
[ERROR] SoapyLogAcceptor::shutdown(tcp://192.168.2.23:55132) FAIL: SoapyRPCPacker::send() FAIL: send() [32: Broken pipe]
[ERROR] SoapyRemote::find(tcp://192.168.2.23:55132) -- transact FAIL: SoapyRPCUnpacker::recv(header) FAIL: 
No devices found!
```

Strykar, 2019-01-24:
@SDRplay Perhaps, it might be helpful to send an SDRPlay to @guruofquality

guruofquality, 2019-01-24:
> if I have 1 device connected locally, start CubicSDR, start streaming

Just checking, does locally mean CubicSDR is talking to SoapySDRPlay directly, or through the SoapyRemote server on "localhost"?

> @guruofquality Josh, I thought you had a RSP - if not, drop me an email - thanks, Andy

Cool, thx, I will email you later

SDRplay, 2019-01-24:
Not using SoapyRemote at all, just 1 RSP connected to the local machine, running CubicSDR and SoapySDRUtil locally. Is this issue just when SoapyRemote is involved?

guruofquality, 2019-02-07:
Hopefully this PR resolves the issue: https://github.com/pothosware/SoapySDRPlay/pull/47
This branch: https://github.com/pothosware/SoapySDRPlay/tree/sdrplay_work

Please give it a try, thx

fmoessbauer, 2019-02-09:
Hi @guruofquality ,

thanks for this fix. I tested it and it works like charm.

closed 2019-02-09 by guruofquality