pothosware/SoapySDRPlay2 #50: SoapySDRUtil probe freeze RSP1

dropdrop20, 2019-03-15:
Running probe multiple times will eventually block the my RSP1. Environment
is Raspberry PI3.

The only chance is to reboot the Raspberry.

Library mir:
root@xxxxx~ # ls -l /usr/local/lib/libmirsdrapi-rsp.so.2.13
-rw-r--r-- 1 root staff 183188 Mar 15 11:00 /usr/local/lib/libmirsdrapi-rsp.so.2.13

root@xxxxx:~ # SoapySDRUtil --probe="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=18070ED596
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=255

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 48] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 9] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

root@xxxxx~ #
root@xxxxxx:~ # SoapySDRUtil --probe="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay
Error probing device: map::at
root@xxxxx:~ # SoapySDRUtil --probe="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay

^C
root@xxxx:~ #

guruofquality, 2019-03-17:
Looks like one of those runs actually caused a crash before it got into this lock up state. FWIW, I didnt see this on my PC, so it might be something related to the driver on ARM specifically.

I know this is terribly annoying, but can you comment out the exception handling here: https://github.com/pothosware/SoapySDR/blob/master/apps/SoapySDRUtil.cpp#L192 and get a traceback from when it throws the first `Error probing device: map::at`?

Alternatively for the freeze up, throw in a tone of prints into SoapySDRPlay: https://github.com/pothosware/SoapySDRPlay/blob/master/Settings.cpp and lets find out what mirics API call is locked up, that should help @SDRplay look at the problem closer.

dropdrop20, 2019-03-18:
I've open a support case to SDRPlay website. Changing Raspberry was a good option. From RPI3 to my old RPI2. Anyway the rtl_airband is still stopping after few minutes.

Which is not happpening with my 8 euro junk RTL-SDR. I'll do experiments and keep 
you informed. Indeed incredible anoying.

SoapySDR: device 'driver=sdrplay': using native sample format 'CS16' (fullScale=32767.0)
RTLSDR-Airband version v3.0.1 starting
SoapySDR: device 'driver=sdrplay': sample rate set to 3000000 sps
SoapySDR: device 'driver=sdrplay': gain set to 25.0 dB
SoapySDR: device 'driver=sdrplay' initialized
[INFO] Using format CS16.
SoapySDR: device 'driver=sdrplay' started
Writing to /root/audio/fm146.2125_20190318_08.mp3
Writing to /root/audio/fm146.2250_20190318_08.mp3
Writing to /root/audio/fm146.2375_20190318_08.mp3
Writing to /root/audio/fm146.2000_20190318_08.mp3
Writing to /root/audio/fm146.1875_20190318_08.mp3
Writing to /root/audio/fm146.0250_20190318_08.mp3
Writing to /root/audio/fm146.0375_20190318_08.mp3
Writing to /root/audio/fm146.0500_20190318_08.mp3
Writing to /root/audio/fm146.0625_20190318_08.mp3
Writing to /root/audio/fm146.0750_20190318_08.mp3
Writing to /root/audio/fm146.0875_20190318_08.mp3
Writing to /root/audio/fm146.1000_20190318_08.mp3
Writing to /root/audio/fm146.1125_20190318_08.mp3
Writing to /root/audio/fm146.1250_20190318_08.mp3
Writing to /root/audio/fm146.2750_20190318_08.mp3
Writing to /root/audio/fm146.2875_20190318_08.mp3
Writing to /root/audio/fm146.1375_20190318_08.mp3
Writing to /root/audio/fm146.1750_20190318_08.mp3
Writing to /root/audio/fm146.2500_20190318_08.mp3
Writing to /root/audio/fm146.2625_20190318_08.mp3
Writing to /root/audio/fm146.0000_20190318_08.mp3
Writing to /root/audio/fm146.0125_20190318_08.mp3
Writing to /root/audio/fm146.1500_20190318_08.mp3
Writing to /root/audio/fm146.1625_20190318_08.mp3
OSoapySDR device 'driver=sdrplay': readStream failed: OVERFLOW, disabling
All receivers failed, exiting
Freeing GPU memory
Cleaning up
Input threads closed

jazzkutya, 2019-04-10:
> I know this is terribly annoying, but can you comment out the exception handling here: https://github.com/pothosware/SoapySDR/blob/master/apps/SoapySDRUtil.cpp#L192 and get a traceback from when it throws the first `Error probing device: map::at`?

Hi,

I don't have the freezes, and I am on x86-64 (ubuntu 14.04lts), but I also get "Error probing device: map::at". I always get this, not just ocassionally. I can't use SoapySDR due to this.
Here is output with the try and whole catch block uncommented:

 Probe device 
 terminate called after throwing an instance of 'std::out_of_range'
   what():  map::at
 Aborted (core dumped)

despite what it says about a core dump, I do not have a core dump file because ubuntu. I will post it if I find it or succeed in producing it. How could I get a proper exception with stack traces and stuff?

jazzkutya, 2019-04-10:
```
(gdb) run --probe
Starting program: /usr/local/bin/SoapySDRUtil --probe
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[New Thread 0x7ffff3e03700 (LWP 25327)]
[New Thread 0x7ffff3602700 (LWP 25328)]
[Thread 0x7ffff3e03700 (LWP 25327) exited]
[New Thread 0x7ffff3e03700 (LWP 25329)]
[Thread 0x7ffff3e03700 (LWP 25329) exited]
[Thread 0x7ffff3602700 (LWP 25328) exited]
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at

Program received signal SIGABRT, Aborted.
0x00007ffff6fbdc37 in raise () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) bt
#0  0x00007ffff6fbdc37 in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff6fc1028 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007ffff77e4535 in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff77e26d6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff77e2703 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff77e26c7 in std::rethrow_exception(std::__exception_ptr::exception_ptr) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7b6ba97 in std::__basic_future<SoapySDR::Device*>::_M_get_result (this=0x7fffffffd6b0) at /usr/include/c++/4.8/future:598
#7  0x00007ffff7b683cc in std::shared_future<SoapySDR::Device*>::get (this=0x7fffffffd6b0) at /usr/include/c++/4.8/future:804
#8  0x00007ffff7b5d90f in SoapySDR::Device::make (inputArgs=...) at /home/vrm/sw/src/SoapySDR-soapy-sdr-0.7.1/lib/Factory.cpp:192
#9  0x00007ffff7b5db56 in SoapySDR::Device::make (args=...) at /home/vrm/sw/src/SoapySDR-soapy-sdr-0.7.1/lib/Factory.cpp:201
#10 0x0000000000413571 in probeDevice (argStr=...) at /home/vrm/sw/src/SoapySDR-soapy-sdr-0.7.1/apps/SoapySDRUtil.cpp:200
#11 0x0000000000413a1c in main (argc=2, argv=0x7fffffffda08) at /home/vrm/sw/src/SoapySDR-soapy-sdr-0.7.1/apps/SoapySDRUtil.cpp:320
(gdb)
```

jazzkutya, 2019-04-10:
"Error probing device: map::at" happens when findSDRPlay() does not find any available devices, so this is probably just the side effect of the root problem. The freezing is probably also a side effect as it happens after the device is stuck in an unavailable state.

guruofquality, 2019-04-11:
Looks like we resolved the map at issue, it was no devices found, its reported better now. I guess any lock up has to be taken up with SDRPlay drivers. Thanks! Reopen if something comes up

closed 2019-04-11 by guruofquality