pothosware/SoapySDRPlay2 #54: SDRPlay version upgrade did change library names

metssigadus, 2019-07-27:
I am informing you that SDRPlay has changed their naming for critical includes while upgrading their API version from SDRplay_RSP_API-Linux-2.13.1.run to SDRplay_RSP_API-Linux-3.01.beta1.run

As the result, SoapySDRPlay is not compiling anymore for SDRPlay b/c the "requirements are not met". The issue: the warning on the git page saying "Get SDR Play driver binaries 'API/HW driver v2.x' (not v3.x)" **remains invisible for people just cloning your code**.

Could it plz be possible to amend the SoapySDRPlay with a **compile time warning** against the 3.0.1 version?! As just a sysadm I am lacking the qualifications to tell you whether just the naming scheme have been changed or were some deeper incompatibilities incorporated. Below are the visible differences, the excerpts from the relevant install_lib files (of SDRPlay API):

**ver 2.13.1**:
echo "Install /usr/local/lib/libmirsdrapi-rsp.so.2.13"
sudo cp ${ARCH}/libmirsdrapi-rsp.so.${VERS} /usr/local/lib/.
sudo chmod 644 /usr/local/lib/libmirsdrapi-rsp.so.${VERS}
sudo ln -s /usr/local/lib/libmirsdrapi-rsp.so.${VERS} /usr/local/lib/libmirsdrapi-rsp.so.2
sudo ln -s /usr/local/lib/libmirsdrapi-rsp.so.2 /usr/local/lib/libmirsdrapi-rsp.so
[..]
echo "Install /usr/local/include/mirsdrapi-rsp.h"
sudo cp mirsdrapi-rsp.h /usr/local/include/.
sudo chmod 644 /usr/local/include/mirsdrapi-rsp.h

**ver 3.0.1**
export VERS="3.01"
export MAJVERS="3"
[..]
echo -n "Installing /usr/local/lib/libsdrplay_api.so.${VERS}..."
sudo rm -f /usr/local/lib/libsdrplay_api.so.${VERS}
sudo cp -f ${ARCH}/libsdrplay_api.so.${VERS} /usr/local/lib/.
sudo chmod 644 /usr/local/lib/libsdrplay_api.so.${VERS}
sudo rm -f /usr/local/lib/libsdrplay_api.so.${MAJVERS}
sudo ln -s /usr/local/lib/libsdrplay_api.so.${VERS} /usr/local/lib/libsdrplay_api.so.${MAJVERS}
sudo rm -f /usr/local/lib/libsdrplay_api.so
sudo ln -s /usr/local/lib/libsdrplay_api.so.${MAJVERS} /usr/local/lib/libsdrplay_api.so
echo "Done"
[..]
echo -n "Installing header files in /usr/local/include..."
sudo cp -f inc/sdrplay_api*.h /usr/local/include/.
sudo chmod 644 /usr/local/include/sdrplay_api*.h
echo "Done"

SDRplay, 2019-07-27:
The API 3.x and API 2.x are fundamentally different and you can't just recompile SoapySDRPlay to support 3.x

One of our customers has made a version of SoapySDRPlay on our forum that is compatible with API 3.x (note the API 3.x is currently fully released only on Windows and released in beta form for Linux.

Hope that helps

SDRplay

SDRplay, 2019-12-15:
api3 branch now supports API 3.06

closed 2019-12-15 by SDRplay

Paula-Bean, 2020-09-14:
Since SDRPlay (the company) now has v3 API as default (trying to download the v2 driver even gives a 404 not found error, seems like they have deprecated that driver), it might be a good idea to merge 'api3' to master. My first try to compile resulted in a 'SDRPlay development files not found' and it was not until after a 'git checkout api3' that SoapySDRPlay would compile succesfully. I'm on Ubuntu 18.04.5 LTS.