pothosware/SoapySDRPlay2 #56: SoapySDRPlay on Windows 10

swandhams, 2019-08-27:
Trying to install SoapySDR on a Windows 10 PC with an SDRPlay RSP2.
(Plan is, once it works locally, to use SoapySDR Remote to connect  to the RSP2 across the LAN.)
Using Cmake GUI, I have configured SoapySDR, but when configuring SoapySDRPlay, I get the following ...

"CMake Error at CMakeLists.txt:7 (find_package):
  Could not find a configuration file for package "SoapySDR" that is
  compatible with requested version "0.4.0".
  The following configuration files were considered but not accepted:
    C:/_SoapySDR_2/src/SoapySDR-master/cmake/Modules/SoapySDRConfig.cmake, version: unknown
"
Tried installing PothosSDR but the necessary files don't appear there either.

I'm a bit out-of-my-league so any help is appreciated.

swandhams, 2019-08-27:
> Typically the trick is to tell cmake how to find the installed project files, you can do this two ways:
>
> * Set the CMAKE_INSTALL_PREFIX to be the same on used to install SoapySDR. Example C:Program Files/PothosSDR
> * Or set SoapySDR_DIR to the the directory that contains the installed SoapySDRConfig.cmake Example: C:Program Files/PothosSDR/cmake

Thanks Josh, 

That's a great help - I'll try that in the morning. 

(11.30pm here) 

Much appreciated. 

Tim 

(Sampford) 

 

Links:
------
[1]
https://github.com/pothosware/SoapySDRPlay/issues/56?email_source=notifications&amp;email_token=ANASZFOBJWRJM4F72FUCEKLQGWPV7A5CNFSM4IQN3T2KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD5JH4JY#issuecomment-525499943
[2]
https://github.com/

vsonnier, 2019-08-28:
Hello @swandhams, you can try to follow the CubicSDR windows buid guide here : https://github.com/cjcliffe/CubicSDR/wiki/Build-Windows that describes how to build both SoapySDR and  SoapySDRPlay.

swandhams, 2019-08-28:
Thank you, both.  I am making progress, but lack of experience means it takes a long time to learn the basics.  Good to know the Win10+RSP2 is feasible - is that with the RSP2 connected locally, or over a remote link, as I need to add SoapySDR Remote later.

SDRplay, 2019-12-15:
The RSPs can be accessed locally or via SoapyRemote.

closed 2019-12-15 by SDRplay