pothosware/SoapySDRPlay2 #57: Cannot build SoapySDRPlay on Ubuntu 18.04 with packaged binaries

ccera-astro, 2019-10-08:
Trying to build SoapySDRplay on Ubuntu 18.04, using the packaged binaries for SoapySDR and friends.

Initially, it couldn't find:

SoapySDRConfig.cmake

So, grabbed the latest source package from git and pointed Cmake at that file in the source package.

That yielded:

CMake Error at CMakeLists.txt:7 (find_package):
  Could not find a configuration file for package "SoapySDR" that is
  compatible with requested version "0.4.0".

  The following configuration files were considered but not accepted:

    /home/mleech/SoapySDR/cmake/Modules/SoapySDRConfig.cmake, version: unknown

Which means that the SoapySDRPlay package needs a version of SoapySDR that is older than 
  that in the PPAs or in the distro packages.

What do, what do?

SDRplay, 2019-10-08:
If you've got SoapySDR version 0.4, that's pretty old - I would recommend getting the latest version of SoapySDR and SoapySDRPlay - they will build in minutes (build them in that order).

guruofquality, 2019-10-08:
@ccera-astro did you install the development package? You may want to read this https://github.com/pothosware/SoapySDR/wiki/ConfigGuide See "Attention debian/ubuntu/PPA users"

> compatible with requested version "0.4.0"

This is just the minimum compatible version thats listed in SoapySDRPlay

ccera-astro, 2019-10-08:
Well, it doesn't appear to be on the system at all.  Hmmm.   I shall 
have to try this again.

Certainly, there are no "-devel" or "-dev" packages in the standard 
Ubuntu packages--do the PPAs install the "-dev[el]" packages?

guruofquality, 2019-10-08:
> Certainly, there are no "-devel" or "-dev" packages in the standard
Ubuntu packages--do the PPAs install the "-dev[el]" packages?

Not sure what you mean the -dev packages are standard for basically and and all C/C++ projects packaged in debian/ubuntu systems. You can see those exact files (ex SoapySDRConfig.cmake) are packaged in ubuntu mainline: https://packages.ubuntu.com/disco/amd64/libsoapysdr-dev/filelist If the PPA happens to have a newer version, the same files should still be present.

> /home/mleech/SoapySDR/cmake/Modules/SoapySDRConfig.cmake, version: unknown

Did you manually tell cmake to look in this directory or was this automatic? if this is an issue, I cmake considers a certain structure in the home directory to be part of its search path. So even though this looks like its a source check out. So if installing the development package doesnt fix this particular error I would move the git clone to a subdirectory. -- Though if you use the development packages from the apt repo, you shouldnt need the source checkout obviously.

ccera-astro, 2019-10-08:
I manually told cmake to look there, using a source checkout of latest 
SoapySDR.   A "find" on the system produced no evidence that the
   relevant .cmake files were anywhere on it, proor to installing the 
source checkout.

So, I now wonder about the repos my ubuntu instance is using--might be 
something fishy about them--this is on a AtomicPi, which comes with
   Ubuntu 18.04 pre-installed.

ccera-astro, 2019-10-09:
This was, after retracing my steps, PEBKAC, and can be closed.

closed 2019-10-09 by ccera-astro