pothosware/SoapySDRPlay2 #60: Override SoapySDR::Device::setGain to account for SDRPlay gains

dlaw, 2019-11-03:
SoapySDR assumes that gains are additive in dB. When `Device::setGain` is called *without* a name, a handy algorithm automatically distributes the requested gain between all available gain stages:
https://github.com/pothosware/SoapySDR/blob/master/lib/Device.cpp#L281

Unfortunately, SDRPlay gains are unconventional. For the RSP1A, there are two gains exposed via SoapySDR, called "RFGR" and "IFGR".
1. "IFGR" stands for "intermediate frequency gain reduction". It takes a value between 20 and 59. Contrary to what SoapySDR expects, the greatest amplification is obtained with the *smallest* numerical value. The IFGR has units of dB.
2. "RFGR" stands for "radio frequency gain reduction". It takes a value between 0 and 9. As with the IFGR, the greatest amplification is obtained with the smallest value. In addition, the RFGR does not have units of dB! The actual gain reduction is nonlinear and frequency-dependent, but tends to be about 7 dB per count.
    (a) Note that RFGR is also settable via the ["rxgain_sel" menu item](https://github.com/pothosware/SoapySDRPlay/blob/master/Settings.cpp#L799).
    (b) RFGR attenuation tables in dB are available in the [SDRPlay API documentation](https://www.sdrplay.com/docs/SDRplay_SDR_API_Specification.pdf), section 5.3.
    (c) Certain sections of the SDRPlay API and technical documents refer to the RFGR as the LNA state. I am not sure which terminology is preferred.

When we put these unconventional gain definitions together with SoapySDR's default gain distribution algorithm, disaster follows. Nearly any call to
`void SoapySDR::Device::setGain(const int, const size_t, const double)`
will force the RFGR to 9 (more than 60 dB of attenuation!) before allocating the remainder to the IFGR. Now you might argue that SDRPlay users should always set their gains by name -- but unfortunately, certain unenlightened applications do not expose the individual named gains, and there may be no way to avoid a call to the automatic gain distribution algorithm.

I believe the simplest solution is for SoapySDRPlay to override the default SoapySDR implementation of void `SoapySDR::Device::setGain(const int, const size_t, const double)`. Rather than attempting to distribute gain between stages, the requested gain should be assigned directly to the IFGR and the RFGR should not be changed. For consistency, we will also override `SoapySDR::Device::getGainRange(const int dir, const size_t channel)` to exclude the RFGR.

We can also consider an alternative solution, which would bring SoapySDRPlay in line with the SoapySDR abstraction model at the cost of breaking some existing configurations. This solution would be:
* Remove RFGR from the list of available gains. It will remain available through the "rfgain_sel" menu item. (Note that SDRPlay's automatic gain control algorithm works in much the same way -- it varies the IFGR but does not touch the RFGR.)
* Invert the sign of IFGR in SoapySDR, so that it spans from -20 to -59 rather than +20 to +59. This will match the SoapySDR sign convention of a bigger number representing more amplification, while preserving a clear correspondence with the numerical IFGR values present in SDRPlay documentation.

Please note that I am using an SDRPlay RSP1A; the gain definitions and ranges are a bit different for the RSP1 and RSP2 (although on a high level, the same issue is present).

I am very new to SoapySDR (and SDR in general, for that matter). I would be happy to send in a pull request, but first I'd appreciate some feedback from the maintainers regarding the preferred approach.

guruofquality, 2019-11-03:
> I am very new to SoapySDR (and SDR in general, for that matter). I would be happy to send in a pull request, but first I'd appreciate some feedback from the maintainers regarding the preferred approach.

Sounds great, thats exactly what overriding setGain is for. A PR would be very helpful, and gives us something specific to discuss.

dlaw, 2019-11-04:
@guruofquality Thanks for the response! I've opened #61 with the larger set of changes to bring SoapySDRPlay in line with the SoapySDR gain abstraction. I think this is the best long-term solution.

A simpler change would be to keep the non-standard RFGR and IFGR, and override setGain to ensure that RFGR is never changed. I think this solution is more confusing in the long term, and it's not consistent with the behavior of other SoapySDR hardware interfaces.
https://github.com/pothosware/SoapySDRPlay/compare/master...dlaw:master

The argument I see for the simpler solution would be concern over breaking existing configurations. But if you agree, I'd prefer to move ahead with the changes as sketched out in #61.