pothosware/SoapySDRPlay2 #63: No devices found! driver=sdrplay

pg8m, 2020-02-29:
I can get it to work,  ik keep getting "no sdrplay device found".  Hoppefully i included all information needed to pinpoint the issue, if not say so

user@machine:~$ cat /etc/os-release
NAME="Linux Mint"
VERSION="19.3 (Tricia)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 19.3"
VERSION_ID="19.3"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.ubuntu.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=tricia
UBUNTU_CODENAME=bionic

user@machine:~$ dmesg
[ 6372.934848] usb 1-3: USB disconnect, device number 3
[ 6390.296634] usb 1-3: new high-speed USB device number 5 using xhci_hcd
[ 6390.445657] usb 1-3: New USB device found, idVendor=1df7, idProduct=2500, bcdDevice= 2.00
[ 6390.445659] usb 1-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 6390.446567] msi2500 1-3:1.0: Registered as swradio0
[ 6390.446568] msi2500 1-3:1.0: SDR API is still slightly experimental and functionality changes may follow

user@machine:~$ lsusb
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 004: ID 04a9:190f Canon, Inc. 
Bus 007 Device 006: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Bus 007 Device 005: ID 08bb:29b3 Texas Instruments PCM2903B Audio CODEC
Bus 007 Device 003: ID 0424:2512 Standard Microsystems Corp. USB 2.0 Hub
Bus 007 Device 002: ID 046d:c084 Logitech, Inc. 
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 2109:0813 VIA Labs, Inc. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0b05:18f3 ASUSTek Computer, Inc. 
Bus 003 Device 002: ID 2109:2813 VIA Labs, Inc. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 058f:6364 Alcor Micro Corp. AU6477 Card Reader Controller
Bus 001 Device 005: ID 1df7:2500  
Bus 001 Device 002: ID 045e:07f8 Microsoft Corp. Wired Keyboard 600 (model 1576)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

user@machine:~$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0-14ec39e)
Available factories... sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

user@machine:~$ SoapySDRUtil --check=sdrplay
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Loading modules... done
Checking driver 'sdrplay'... PRESENT

user@machine:~$ SoapySDRUtil --find="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

No devices found! driver=sdrplay

user@machine:~$ SoapySDRUtil --probe="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay
Error probing device: no sdrplay device found

pg8m, 2020-02-29:
****

closed 2020-02-29 by pg8m

SDRplay, 2020-02-29:
Looks like you've got a RSP1 - it could be used by another module. You will need to blacklist those modules for this to work. Add the following lines to the /etc/modprobe.d/blacklist.conf file

blacklist sdr_msi3101
blacklist msi001
blacklist msi2500

Then reboot and try the SoapySDRUtil commands again.

pg8m, 2020-02-29:
It is an RPS1 and blacklisting helped, thanks for the extreme quick response. 

This  is the output i get now,  looks good.   73 Marcel de PG8M

**user@machine:~$ SoapySDRUtil --probe="driver=sdrplay"**
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=RSP1
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=1
  serial=B0001P0003

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=1, type=string, options=(0, 1, 2, 3)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 42] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 3] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz