pothosware/SoapySDRPlay2 #64: Raspbian Buster cmake missing dev.files

catalinalb, 2020-03-17:
Hi, I have raspian buster running on RPi 3 B+. 

I have done these steps:
sudo apt install cmake g++ libpython-dev python-numpy swig git
git clone https://github.com/pothosware/SoapySDR.git
cd SoapySDR
mkdir build
cd build
cmake ..
make -j4
sudo make install
sudo ldconfig
chmod 777 SDRplay_RSP_API-RPi-2.11.1.run
./SDRplay_RSP_API-RPi-2.11.1.run
git clone https://github.com/pothosware/SoapySDRPlay.git
cd SoapySDRPlay
mkdir build
cd build
cmake ..

and here he told me: 

CMake Error at CMakeLists.txt:16 (message):
  SDRPlay development files not found...

Any ideeas?
Thank you

catalinalb, 2020-03-17:
Solved.

I had the wrong API drivers. I should have used the older version (SDRplay_RSP_API-RPi-2.13.1.run). Now it's working.

catalinalb, 2020-03-17:
yeah.... installed. Still unable to run my RSP1a into Raspian Buster. (not found under Cubicsdr menu)

SDRplay, 2020-05-28:
Can't replicate this - RSPs work fine with SoapySDRPlay in our RPi images (https://www.sdrplay.com/raspberry-pi-images) - maybe worth checking those out.

closed 2020-05-28 by SDRplay