pothosware/SoapySDRPlay2 #67: Issues compiling

afromanDAN, 2020-06-15:
I'm on arch linux and I'm getting this compile error.

``/usr/bin/ld: CMakeFiles/sdrPlaySupport.dir/Streaming.cpp.o: in function `SoapySDRPlay::acquireReadBuffer(SoapySDR::Stream*, unsigned long&, void const**, int&, long long&, long)':
Streaming.cpp:(.text+0x872): undefined reference to `pthread_cond_clockwait'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sdrPlaySupport.dir/build.make:151: libsdrPlaySupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:96: CMakeFiles/sdrPlaySupport.dir/all] Error 2
make: *** [Makefile:150: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...``

guruofquality, 2020-06-17:
Just a guess, but it might be a missing pthread call from a recent change in the driver

Can you try adding `-pthread` to the list of options for "LIBRARIES"?

* https://github.com/pothosware/SoapySDRPlay/blob/master/CMakeLists.txt#L48

Should look like
```
LIBRARIES
        ${LIBSDRPLAY_LIBRARIES}
        -pthread
```

afromanDAN, 2020-06-18:
Yep, that did the trick! Thank you!

closed 2020-06-18 by afromanDAN