pothosware/SoapySDRPlay2 #71: RSP1A and rf notch

iwiq, 2020-12-03:
Greetings everyone.

Could you, please, help me with enable rfnotch and biasT.
Device arguments in osmocom block: "soapy=0,driver=sdrplay".

When I use RSP1A in GNUradio and other software via gr-osmosdr I see strong signal 88-108MHz and DC voltage from SMA connector is zero. But Rfnotch works very well in CubicSDR.
**Maybe need to add something to "soapy=0,driver=sdrplay" such as "biasT_ctrl=1" ?**

righthalfplane, 2020-12-15:
SdrGlut uses SoapySDR and there you set rfnotch_cntrl to true to turn on the notch filter. If osmocom is passing the options to soapy, the option name would also be rfnotch_cntrl and rfnotch_cntrl=1 or rfnotch_cntrl=true should work.

fventuri, 2020-12-16:
Thanks @righthalfplane
I hadn't realized @iwiq asked this question here until I received the email alert about your comment.
We had a similar discussion about the 'rfnotch_cntrl' argument here: https://github.com/SDRplay/SoapySDRPlay/issues/21

Franco

ChrisVine, 2021-11-21:
@fventuri The fix you applied to SoapySDRPlay3 to deal with this seems to work fine for me with SoapySDRPlay2.  Why not apply it to SoapySDRPlay2, as all the option handling in the writeSetting() function is already there - patch below?  (As an aside, strictly speaking you ought to include the <utility> header because that is where std::pair is found but the same applies to SoapySDRPlay3 and the patch below just reproduces what you have done in that repository.)

```
--- a/Settings.cpp
+++ b/Settings.cpp
@@ -69,6 +69,17 @@ SoapySDRPlay::SoapySDRPlay(const SoapySDR::Kwargs &args)
     bwMode = mir_sdr_BW_1_536;
     gRdB = 40;
     lnaState = (hwVer == 2 || hwVer == 3 || hwVer > 253)? 4: 1;
+    
+    // process additional device string arguments
+    for (std::pair<std::string, std::string> arg : args) {
+        // ignore 'driver', 'label', 'mode', 'serial', and 'soapy'
+        if (arg.first == "driver" || arg.first == "label" ||
+	    arg.first == "mode" || arg.first == "serial" ||
+	    arg.first == "soapy") {
+	  continue;
+	}
+	writeSetting(arg.first, arg.second);
+    }
 
     //this may change later according to format
     shortsPerWord = 1;
```

fventuri, 2021-11-21:
@ChrisVine - thanks for the useful suggestion!

I just created a new branch called `add_writeSetting_to_constructor` with your proposed change. I just moved a few lines down in the code, since there were lines like `biasTen = 0;` afterwards, and I was afraid they would have overwritten the user selection.

Also, I was able to build it it here without adding #include &lt;utility&gt;, so I suspect it is already included in one of the other standard header files.

Anyhow, please give it a try with version 2 of the SDRplay API (I just built it here, but I didn't install it since I run version 3), and let me know how it goes. If it works for you, I'll merge this change into the `master` branch.

Franco

ChrisVine, 2021-11-21:
I have tested setting options for a RSP1A device with the proprietary sdrplay 2.13.1 driver and gnuradio-3.9.4.0, using the add_writeSettings_to_constructor branch, as follows:

1. gnuradio-companion flowcharts using the built-in soapy source block continue to work correctly -- you can set options using a driver string both with and without the patch (there is no change in behaviour with the patch applied, which is what is wanted)
2. gnuradio-companion flowcharts using a gr-osmosdr source with the soapy backend (that is, a 'soapy=0,driver=sdrplay' driver string) now accept additional options such as 'rfnotch_ctrl=true' correctly for the first time.
3. gqrx with a driver string of 'soapy=0, driver=sdrplay' now also accepts additional options such as 'rfnotch_ctrl=true' correctly for the first time.

So all seems good.

fventuri, 2021-11-21:
Thanks for the thorough validation Chris; I just merged this change into `master` (https://github.com/pothosware/SoapySDRPlay2/pull/72)

Franco