pothosware/SoapySDRPlay2 #73: MAC OS timing issue - RSP1A not found on first attemp , second ok ?

aneitzel, 2022-08-19:
Hi all,

found a problem with SoapySDRplay on MAC. Have tested a complete fresh instal of OS (Catalina 10.15.7) and all prereq + SDRplay via PORTS. After boot OS and check SDRplay hw with SoapySDRUtil -find, it dont find the RSP1A on first attemp. After rerun the same command it works!

alpha@iMAC ~ % SoapySDRUtil -find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

libusb: error [darwin_claim_interface] interface not found
**No devices found!** 
alpha@iMAC ~ % SoapySDRUtil -find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  driver = sdrplay
  label = SDRplay Dev0 RSP1A 1807082793
  serial = 18XXXXXXXX

---------------------------------------------------

Any idea whats wrong? 

---------------------------------------------------

alpha@iMAC ~ % SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-unknown
API Version: v0.8.0
ABI Version: v0.8
Install root: /opt/local
Search path:  /opt/local/lib/SoapySDR/modules0.8
Module found: /opt/local/lib/SoapySDR/modules0.8/libLMS7Support.so    (20.10.0)
Module found: /opt/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0)
Available factories... lime, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
alpha@iMAC ~ % SoapySDRUtil --probe=sdrplay
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device sdrplay

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=RSP1A
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=255
  serial=180XXXXXXXXX

fventuri, 2022-08-19:
@aneitzel - I don't have a Mac (just Linux and Windows here), but I would try to figure out why you are getting that message the first time:

> libusb: error [darwin_claim_interface] interface not found
No devices found!

On Linux I would use the command `lsusb` before the first time to see the USB devices and interfaces seen by the kernel right after booting it; there's probably a similar command for MacOS that might help you further diagnose this issue.

Hope it helps,
Franco

aneitzel, 2022-08-20:
on MAC there is lsbutils available via PORTS also, seems my USB RSP1A is recognized correctly after boot. This is what i got via lsusb:

alpha@iMAC ~ % lsusb 
Bus 020 Device 010: ID 05ac:0250 Apple, Inc. Aluminium Keyboard (ISO)
**Bus 020 Device 007: ID 1df7:3000 SDRplay RSP1a**
Bus 020 Device 005: ID 05ac:8511 Apple, Inc. FaceTime HD Camera (Built-in)
Bus 020 Device 004: ID 05ac:1006 Apple, Inc. Hub in Aluminum Keyboard
Bus 020 Device 003: ID 08bb:2902 Texas Instruments PCM2902 Audio Codec

But i need to fire up one time a "SoapySDRUtil --find" to enable soapy and SDR software using it to find it.

From my point of view it seems to be a issue somethere between OS USB and soapy and or SDRplay code :-(
Any other user who run into the same?

fventuri, 2022-08-20:
@aneitzel perhaps this is what is going on:
- right after the boot the RSP1A device is still in some sort of non-initialized mode
- the first time you run `SoapySDRUtil`, the SDRplay API downloads the proper firmware into the device and initializes the device correctly
- the second time you run `SoapySDRUtil` the USB RSP1A is fully initialized and it responds correctly to the USB requests

Since you have `lsusb` there, you can see the difference in the USB device configuration by running `lsusb -v -d 1df7:3000`, and looking at the line with `bcdDevice`; right after boot this is what I see here on Linux:
```
  bcdDevice            2.00
```
and after the first time I run `SoapySDRUtil` it becomes (your number might be different):
```
  bcdDevice            2.07
```

BTW I see you are still running an old version (2.13) of the SDRplay API - it is several years old, and I think it is no longer supported by SDRplay.
You may want to try the current version (3.07 for Linux and Mac); the SoapySDRPlay3 module is available here: https://github.com/pothosware/SoapySDRPlay3 and the build instructions are here: https://github.com/pothosware/SoapySDRPlay3/wiki (I am not sure if there are packages ready to install for Macs in homebrew or macports).

Franco